jayofelony/pwnagotchi-bullseye #1: RPi4 Blind on boot

renewedbullet, 2023-08-23:
<html>
<body>

[2023-08-23 19:33:11,698] | [INFO] | [ai] ! sad_num_epochs: 15 -> 24
-- | -- | --
[2023-08-23 19:33:11,698] | [INFO] | [ai] ! channels: [5, 6, 7, 10, 40, 44, 48, 52, 60, 64, 104, 108, 112, 120, 124, 128, 132, 136, 140] -> [2, 4, 9, 10, 40, 44, 52, 56, 60, 100, 112, 124, 128, 132, 136, 140]
[2023-08-23 19:33:13,202] | [INFO] | [epoch 8] duration=00:00:11 slept_for=00:00:10 blind=8 sad=0 bored=0 inactive=9 active=0 peers=0 tot_bond=0.00 avg_bond=0.00 hops=0 missed=0 deauths=0 assocs=0 handshakes=0 cpu=40% mem=10% temperature=42C reward=-0.4666666666666667
[2023-08-23 19:33:13,204] | [INFO] | [ai] saving model to /root/brain.nn ...
[2023-08-23 19:33:13,505] | [INFO] | [ai] saving /root/brain.json
[2023-08-23 19:33:21,225] | [INFO] | [ai] setting new policy:
[2023-08-23 19:33:21,227] | [INFO] | [ai] ! min_rssi: -143 -> -79
[2023-08-23 19:33:21,227] | [INFO] | [ai] ! ap_ttl: 386 -> 390
[2023-08-23 19:33:21,228] | [INFO] | [ai] ! sta_ttl: 114 -> 165
[2023-08-23 19:33:21,228] | [INFO] | [ai] ! recon_time: 29 -> 18
[2023-08-23 19:33:21,229] | [INFO] | [ai] ! max_inactive_scale: 3 -> 8
[2023-08-23 19:33:21,230] | [INFO] | [ai] ! hop_recon_time: 18 -> 42
[2023-08-23 19:33:21,230] | [INFO] | [ai] ! min_recon_time: 4 -> 1
[2023-08-23 19:33:21,245] | [INFO] | [ai] ! max_interactions: 1 -> 22
[2023-08-23 19:33:21,245] | [INFO] | [ai] ! max_misses_for_recon: 5 -> 6
[2023-08-23 19:33:21,246] | [INFO] | [ai] ! excited_num_epochs: 21 -> 11
[2023-08-23 19:33:21,247] | [INFO] | [ai] ! bored_num_epochs: 29 -> 21
[2023-08-23 19:33:21,248] | [INFO] | [ai] ! sad_num_epochs: 24 -> 20
[2023-08-23 19:33:21,249] | [INFO] | [ai] ! channels: [2, 4, 9, 10, 40, 44, 52, 56, 60, 100, 112, 124, 128, 132, 136, 140] -> [3, 4, 6, 11, 36, 44, 48, 64, 100, 104, 116, 120, 124, 128, 132, 136]

</body>
</html>

01:01:46 !!! dial tcp: lookup pwngrid-mysql on 8.8.8.8:53: read udp 172.20.10.7:                            52025->8.8.8.8:53: i/o timeout
01:02:17 inf pwngrid v1.10.3 starting in server mode ...
01:02:17 inf connecting to pwngrid-mysql:3306 ...
01:02:18 !!! dial tcp: lookup pwngrid-mysql on 8.8.8.8:53: no such host
01:02:49 inf pwngrid v1.10.3 starting in server mode ...
01:02:49 inf connecting to pwngrid-mysql:3306 ...
01:02:54 !!! dial tcp: lookup pwngrid-mysql on 8.8.8.8:53: no such host
01:03:24 inf pwngrid v1.10.3 starting in server mode ...
01:03:24 inf connecting to pwngrid-mysql:3306 ...
01:03:24 !!! dial tcp: lookup pwngrid-mysql on 8.8.8.8:53: no such host
01:03:54 inf pwngrid v1.10.3 starting in server mode ...
01:03:54 inf connecting to pwngrid-mysql:3306 ...
01:03:54 !!! dial tcp: lookup pwngrid-mysql on 8.8.8.8:53: no such host
01:04:24 inf pwngrid v1.10.3 starting in server mode ...
01:04:24 inf connecting to pwngrid-mysql:3306 ...
01:04:24 !!! dial tcp: lookup pwngrid-mysql on 8.8.8.8:53: no such host

/var/log/pwngrid-peer.log <==
10:41:54 inf pwngrid v1.10.3 starting in peer mode ...
10:42:02 inf /etc/pwnagotchi/id_rsa found
10:42:02 inf started beacon discovery and message routing (0 known peers)
10:42:02 inf peer pwnagotchi@c79022a9ec2a2cd8d01bb3c677e8495b6a827758070f939f719                            e822d51a0d765 signaling is ready
10:42:02 inf pwngrid api starting on 127.0.0.1:8666 ...
10:42:02 inf detected new peer aleda@ebd94ff44b1f762e46400151eed4f01b6a03f1a43c8                            f0f7365d8e937102e9a33 on channel 10
10:42:04 inf peer advertisement enabled
10:42:12 inf pwngrid v1.10.3 starting in peer mode ...
10:42:14 inf /etc/pwnagotchi/id_rsa found
10:42:14 inf interface wlan0mon is down, bringing it up ...
10:42:14 inf started beacon discovery and message routing (1 known peers)
10:42:14 inf peer pwnagotchi@c79022a9ec2a2cd8d01bb3c677e8495b6a827758070f939f719                            e822d51a0d765 signaling is ready
10:42:14 inf pwngrid api starting on 127.0.0.1:8666 ...
10:42:19 inf peer advertisement enabled
10:42:22 inf detected new peer aleda@ebd94ff44b1f762e46400151eed4f01b6a03f1a43c8                            f0f7365d8e937102e9a33 on channel 5
10:49:20 inf pwngrid v1.10.3 starting in peer mode ...
10:49:22 inf /etc/pwnagotchi/id_rsa found
10:49:22 inf interface wlan0mon is down, bringing it up ...
10:49:22 inf started beacon discovery and message routing (1 known peers)
10:49:22 inf peer aleda2@c79022a9ec2a2cd8d01bb3c677e8495b6a827758070f939f719e822                            d51a0d765 signaling is ready
10:49:22 inf pwngrid api starting on 127.0.0.1:8666 ...
10:49:27 inf peer advertisement enabled
10:49:32 inf detected new peer aleda@ebd94ff44b1f762e46400151eed4f01b6a03f1a43c8                            f0f7365d8e937102e9a33 on channel 9
10:53:18 inf pwngrid v1.10.3 starting in peer mode ...
10:53:21 inf /etc/pwnagotchi/id_rsa found
[... 191 lines cut ...]
19:33:11 [INFO] [ai] ! excited_num_epochs: 12 -> 21
19:33:11 [INFO] [ai] ! bored_num_epochs: 14 -> 29
19:33:11 [INFO] [ai] ! sad_num_epochs: 15 -> 24
19:33:11 [INFO] [ai] ! channels: [5, 6, 7, 10, 40, 44, 48, 52, 60, 64, 104, 108, 112, 120, 124, 128, 132, 136, 140] -> [2, 4, 9, 10, 40, 44, 52, 56, 60, 100, 112, 124, 128, 132, 136, 140]
19:33:13 [INFO] [epoch 8] duration=00:00:11 slept_for=00:00:10 blind=8 sad=0 bored=0 inactive=9 active=0 peers=0 tot_bond=0.00 avg_bond=0.00 hops=0 missed=0 deauths=0 assocs=0 handshakes=0 cpu=40% mem=10% temperature=42C reward=-0.4666666666666667
19:33:13 [INFO] [ai] saving model to /root/brain.nn ...
19:33:13 [INFO] [ai] saving /root/brain.json
19:33:21 [INFO] [ai] setting new policy:
19:33:21 [INFO] [ai] ! min_rssi: -143 -> -79
19:33:21 [INFO] [ai] ! ap_ttl: 386 -> 390
19:33:21 [INFO] [ai] ! sta_ttl: 114 -> 165
19:33:21 [INFO] [ai] ! recon_time: 29 -> 18
19:33:21 [INFO] [ai] ! max_inactive_scale: 3 -> 8
19:33:21 [INFO] [ai] ! hop_recon_time: 18 -> 42
19:33:21 [INFO] [ai] ! min_recon_time: 4 -> 1
19:33:21 [INFO] [ai] ! max_interactions: 1 -> 22
19:33:21 [INFO] [ai] ! max_misses_for_recon: 5 -> 6
19:33:21 [INFO] [ai] ! excited_num_epochs: 21 -> 11
19:33:21 [INFO] [ai] ! bored_num_epochs: 29 -> 21
19:33:21 [INFO] [ai] ! sad_num_epochs: 24 -> 20
19:33:21 [INFO] [ai] ! channels: [2, 4, 9, 10, 40, 44, 52, 56, 60, 100, 112, 124, 128, 132, 136, 140] -> [3, 4, 6, 11, 36, 44, 48, 64, 100, 104, 116, 120, 124, 128, 132, 136]
19:34:12 [INFO] [epoch 9] duration=00:00:59 slept_for=00:00:58 blind=9 sad=0 bored=0 inactive=10 active=0 peers=0 tot_bond=0.00 avg_bond=0.00 hops=0 missed=0 deauths=0 assocs=0 handshakes=0 cpu=29% mem=10% temperature=42C reward=-0.47000000000000003
19:34:12 [INFO] [ai] saving model to /root/brain.nn ...
19:34:12 [INFO] [ai] saving /root/brain.json
[image]

renewedbullet, 2023-08-23:
works with disabling onboard wifi and using a alfa dongle

jayofelony, 2023-08-23:
Can you share output of

`sudo airmon-ng --test wlan0mon`

It should say injection is working and x-number of aps found.

renewedbullet, 2023-08-24:
[image]

This was a fresh image too

jayofelony, 2023-08-24:
Oh my bad I meant aireplay-ng instead of airmon-ng

renewedbullet, 2023-08-24:
[image]

renewedbullet, 2023-08-24:
after about 45 minutes it rebooted and went blind again.
[image]

renewedbullet, 2023-08-25:
[image]

jayofelony, 2023-08-25:
This is because of damaged brain, remove /root/brain.nn and reboot

closed 2023-08-26 by jayofelony