jayofelony/pwnagotchi-bullseye #2: Mysql? [BUG]

Dviros, 2023-08-28:
**Describe the bug**
Hi, went through the logs and had this error:
```
12:07:47 inf connecting to pwngrid-mysql:3306 ...
12:07:47 !!! dial tcp: lookup pwngrid-mysql on 8.8.8.8:53: no such host
```

Trying to understand why would it look for a mysql server :D
Appreciate your thoughts on this :) Thanks!

**To Reproduce**
Steps to reproduce the behavior:
1. ...
2. ...
3. ...

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Environment (please complete the following information):**
 - Pwnagotchi version
 - OS version
 - Type of hardware
 - Any additional hardware used

**Additional context**
Add any other context about the problem here.

jayofelony, 2023-08-28:
Yes, it is looking up pwngrid-mysql on the dns server 8.8.8.8, but that hostname does not exist. Nothing to worry about. You can ignore the error. It'll work perfectly with it.

closed 2023-08-28 by jayofelony

do-ki, 2023-08-29:
also encouter this averything seems to work fine.

do-ki, 2023-09-02:
I can confirm  that this error is no longer showing up in the latest version V2.2.1.