jayofelony/pwnagotchi-bullseye #3: [ERROR] Raspberry Pi Zero 2 W

Levi-Michael, 2023-08-28:
after flash 

pi@10.0.0.2's password:
Linux pwnagotchi 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
Last login: Sun Aug 27 23:17:43 2023 from fe80::14ed:e1c6:f3d1:fc9c%usb0

SSH is enabled and the default password for the 'pi' user has not been changed.
This is a security risk - please login as the 'pi' user and type 'passwd' to set a new password.

pi@pwnagotchi:~ $ tail -f /var/log/pwnagotchi.log
FileNotFoundError: [Errno 2] No such file or directory: '/sys/class/leds/led0/brightness'
[2023-08-27 23:36:48,739] [ERROR] [led] error while blinking
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/dist-packages/pwnagotchi/plugins/default/led.py", line 63, in _worker
    self._blink(pattern)
  File "/usr/local/lib/python3.9/dist-packages/pwnagotchi/plugins/default/led.py", line 49, in _blink
    self._led(0)
  File "/usr/local/lib/python3.9/dist-packages/pwnagotchi/plugins/default/led.py", line 40, in _led
    with open(self._led_file, 'wt') as fp:
FileNotFoundError: [Errno 2] No such file or directory: '/sys/class/leds/led0/brightness'
[2023-08-27 23:38:47,249] [ERROR] [led] error while blinking
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/dist-packages/pwnagotchi/plugins/default/led.py", line 63, in _worker
    self._blink(pattern)
  File "/usr/local/lib/python3.9/dist-packages/pwnagotchi/plugins/default/led.py", line 49, in _blink
    self._led(0)
  File "/usr/local/lib/python3.9/dist-packages/pwnagotchi/plugins/default/led.py", line 40, in _led
    with open(self._led_file, 'wt') as fp:
FileNotFoundError: [Errno 2] No such file or directory: '/sys/class/leds/led0/brightness'
[2023-08-27 23:38:47,274] [INFO] [epoch 12] duration=00:02:08 slept_for=00:01:58 blind=12 sad=0 bored=0 inactive=13 active=0 peers=0 tot_bond=0.00 avg_bond=0.00 hops=0 missed=0 deauths=0 assocs=0 handshakes=0 cpu=4% mem=60% temperature=56C reward=-0.47692307692307695
[2023-08-27 23:38:47,278] [ERROR] [led] error while blinking
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/dist-packages/pwnagotchi/plugins/default/led.py", line 63, in _worker
    self._blink(pattern)
  File "/usr/local/lib/python3.9/dist-packages/pwnagotchi/plugins/default/led.py", line 49, in _blink
    self._led(0)
  File "/usr/local/lib/python3.9/dist-packages/pwnagotchi/plugins/default/led.py", line 40, in _led
    with open(self._led_file, 'wt') as fp:
FileNotFoundError: [Errno 2] No such file or directory: '/sys/class/leds/led0/brightness'
[2023-08-27 23:38:57,311] [ERROR] [led] error while blinking
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/dist-packages/pwnagotchi/plugins/default/led.py", line 63, in _worker
    self._blink(pattern)
  File "/usr/local/lib/python3.9/dist-packages/pwnagotchi/plugins/default/led.py", line 49, in _blink
    self._led(0)
  File "/usr/local/lib/python3.9/dist-packages/pwnagotchi/plugins/default/led.py", line 40, in _led
    with open(self._led_file, 'wt') as fp:
FileNotFoundError: [Errno 2] No such file or directory: '/sys/class/leds/led0/brightness'

renewedbullet, 2023-08-28:
turn off LED plugin fixed it for me.

jayofelony, 2023-08-28:
LED is disabled by default in default.toml, if you have an old config delete this line from your config.toml.

closed 2023-08-28 by jayofelony

Levi-Michael, 2023-08-28:
problem was at my config file 
recommended to try default one 
https://github.com/jayofelony/pwnagotchi/blob/pwnagotchi-torch/pwnagotchi/defaults.toml