jayofelony/pwnagotchi-bullseye #4: [BUG] Something "crashes" and then it becomes blind, to fix, just "reboot" and it's up and running until it crashes again.

rsun0525, 2023-08-30:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:

Just let it run

**Expected behavior**

Would expect it to reboot sooner or "fix" the broken service that died

**Screenshots**
Will upload log file so you can see what happens.

**Environment (please complete the following information):**
 - Pwnagotchi version 2.1
 - OS version Linux pwnagotchiz1 6.1.21-v8+ #1642 
 - Type of hardware Raspberry Pi 3B+ with onboard WiFi disabled and external USB WiFi added

**Additional context**
Add any other context about the problem here.
[pwnagotchi.log](https://github.com/jayofelony/pwnagotchi/files/12477849/pwnagotchi.log)

jayofelony, 2023-08-30:
All I see is it lost connection to pwngrid somehow. Seems to fix itself, atleast that's what I get from the log. 

What exactly are seeing it doing? Because I don't know how my image is causing this, because you disabled onboard WiFi and added your own antenna. Did you install the proper drivers for it?

rsun0525, 2023-08-30:
I used a "Panda Wireless PAU09 N600 Dual Band (2.4GHz and 5GHz) Wireless N USB Adapter W/Dual 5dBi Antennas - Windows XP/Vista/7/8/8.1/10/11, Mint, Ubuntu, openSUSE, Fedora, Centos, Kali Linux and Raspbian", so from my understanding the drivers are built in, and everything works great until some weird problem happens and then it turns "blind".  Sorry I'm a novice here, so not sure what's going on, what I can do is turn off grid, and then if/when the next error occurs, then it'll turn blind again.  So you don't see it "stopping"?  When you looked at the log you saw it capturing properly and then suddenly stopping? I don't have this issue with this hardware on other forks (or I should say I have different issues I encounter with the different forks of pwnagotchi :-))

rsun0525, 2023-08-30:
So I turned off rebooted, turned off grid, and now this next thing died, and now it's blind again:

[2023-08-30 16:31:49,319] [INFO] sending association frame to Syracusehome (b0:95:75:37:4d:fa Tp-Link Technologies Co.,Ltd.) on channel 3 [0 clients], -88 dBm...
[2023-08-30 16:31:49,533] [INFO] sending association frame to Syracusehome (b0:95:75:37:40:b6 Tp-Link Technologies Co.,Ltd.) on channel 3 [0 clients], -86 dBm...
[2023-08-30 16:31:49,626] [INFO] sending association frame to <hidden> (b6:95:75:37:4d:fa ) on channel 3 [0 clients], -88 dBm...
[2023-08-30 16:31:49,722] [INFO] sending association frame to <hidden> (b6:95:75:37:40:b6 ) on channel 3 [0 clients], -80 dBm...
[2023-08-30 16:31:49,776] [INFO] waiting for 26s on channel 3 ...
[2023-08-30 16:32:14,985] [ERROR] Task exception was never retrieved
future: <Task finished name='Task-1' coro=<Client.start_websocket() done, defined at /usr/local/lib/python3.9/dist-packages/pwnagotchi/bettercap.py:40> exception=OSError("Multiple exceptions: [Errno 111] Connect call failed ('127.0.0.1', 8081), [Errno 99] Cannot assign requested address")>
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/dist-packages/pwnagotchi/bettercap.py", line 44, in start_websocket
    async with websockets.connect(s, ping_interval=60, ping_timeout=90) as ws:
  File "/usr/local/lib/python3.9/dist-packages/websockets/legacy/client.py", line 637, in __aenter__
    return await self
  File "/usr/local/lib/python3.9/dist-packages/websockets/legacy/client.py", line 655, in __await_impl_timeout__
    return await self.__await_impl__()
  File "/usr/local/lib/python3.9/dist-packages/websockets/legacy/client.py", line 659, in __await_impl__
    _transport, _protocol = await self._create_connection()
  File "/usr/lib/python3.9/asyncio/base_events.py", line 1064, in create_connection
    raise OSError('Multiple exceptions: {}'.format(
OSError: Multiple exceptions: [Errno 111] Connect call failed ('127.0.0.1', 8081), [Errno 99] Cannot assign requested address
[2023-08-30 16:32:15,849] [INFO] CHANNEL 7
[2023-08-30 16:32:15,863] [ERROR] Error while setting channel (HTTPConnectionPool(host='localhost', port=8081): Max retries exceeded with url: /api/session (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x7f34551d60>: Failed to establish a new connection: [Errno 111] Connection refused')))
[2023-08-30 16:32:15,865] [INFO] 3 access points on channel 7
[2023-08-30 16:32:15,903] [INFO] sending association frame to TheHouseNet (f0:2f:74:fa:c4:e1 ASUSTek COMPUTER INC.) on channel 7 [0 clients], -86 dBm...
[2023-08-30 16:32:15,913] [ERROR] HTTPConnectionPool(host='localhost', port=8081): Max retries exceeded with url: /api/session (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x7f345510a0>: Failed to establish a new connection: [Errno 111] Connection refused'))
[2023-08-30 16:32:16,094] [INFO] sending association frame to TheHouseNet (fc:34:97:98:4e:d1 ASUSTek COMPUTER INC.) on channel 7 [0 clients], -80 dBm...
[2023-08-30 16:32:16,109] [ERROR] HTTPConnectionPool(host='localhost', port=8081): Max retries exceeded with url: /api/session (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x7f344bd760>: Failed to establish a new connection: [Errno 111] Connection refused'))
[2023-08-30 16:32:16,188] [INFO] sending association frame to TheHouseNet (fc:34:97:98:54:60 ASUSTek COMPUTER INC.) on channel 7 [0 clients], -88 dBm...
[2023-08-30 16:32:16,196] [ERROR] HTTPConnectionPool(host='localhost', port=8081): Max retries exceeded with url: /api/session (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x7f345510d0>: Failed to establish a new connection: [Errno 111] Connection refused'))
[2023-08-30 16:32:16,239] [INFO] waiting for 26s on channel 3 ...
[2023-08-30 16:32:16,577] [ERROR] [agent:_fetch_stats] self.session: ConnectionError(MaxRetryError("HTTPConnectionPool(host='localhost', port=8081): Max retries exceeded with url: /api/session (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x7f34551b50>: Failed to establish a new connection: [Errno 111] Connection refused'))"))
[2023-08-30 16:32:21,650] [ERROR] [agent:_fetch_stats] self.session: ConnectionError(MaxRetryError("HTTPConnectionPool(host='localhost', port=8081): Max retries exceeded with url: /api/session (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x7f345308b0>: Failed to establish a new connection: [Errno 111] Connection refused'))"))
[2023-08-30 16:32:26,710] [ERROR] [agent:_fetch_stats] self.session: ConnectionError(MaxRetryError("HTTPConnectionPool(host='localhost', port=8081): Max retries exceeded with url: /api/session (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x7f34530c40>: Failed to establish a new connection: [Errno 111] Connection refused'))"))
[2023-08-30 16:32:31,792] [ERROR] [agent:_fetch_stats] self.session: ConnectionError(MaxRetryError("HTTPConnectionPool(host='localhost', port=8081): Max retries exceeded with url: /api/session (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x7f345a2eb0>: Failed to establish a new connection: [Errno 111] Connection refused'))"))
[2023-08-30 16:32:36,893] [ERROR] [agent:_fetch_stats] self.session: ConnectionError(MaxRetryError("HTTPConnectionPool(host='localhost', port=8081): Max retries exceeded with url: /api/session (Caused by NewConnectionEr [... cut ...]
[2023-08-30 16:32:42,561] [ERROR] HTTPConnectionPool(host='localhost', port=8081): Max retries exceeded with url: /api/session (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x7f34551c10>: Failed to establish a new connection: [Errno 111] Connection refused'))
[2023-08-30 16:32:42,600] [INFO] waiting for 26s on channel 3 ...
[2023-08-30 16:33:08,673] [INFO] CHANNEL 100
[2023-08-30 16:33:08,690] [INFO] 2 access points on channel 100
[2023-08-30 16:33:08,777] [INFO] sending association frame to Fios-2YqSz (8a:bd:c5:0a:aa:9a ) on channel 100 [0 clients], -88 dBm...

jayofelony, 2023-08-30:
Yes pwngrid-peers.service needs to be active for Pwnagotchi to be working. It seems as if your overloading the api with so many aps that the connection dies.

I doubt there is much I can do, if you say the antenna is supported then I have to believe you. But I also doubt it is because of my image, since the onboard WiFi works perfectly fine.

jayofelony, 2023-08-30:
Yeah try the onboard wifi, see what happens. If it stays alive than you know the antenna is the culprit. It could be that it's just too much data to handle for your pi3b+.

rsun0525, 2023-08-30:
I'm going to move this microSD card to a 4B with the external USB WiFi adaptor and give it a try (with grid still turned off).   So in your opinion this should stay up for prolonged periods of time?  How long have you run a pwnagotchi before having to reboot it?  Then I'll also setup the 3B+ with the onboard WiFi only and compare.

jayofelony, 2023-08-31:
pwngrid needs to be active for Pwnagotchi to work. So leave that as is.

My Pi02W ran 10hrs30mins, until my powerbank was drained.

rsun0525, 2023-08-31:
So far with the Raspberry Pi 3B+ using the built in WiFi, once in a the wlanmon0 disappears and then I need to reboot to solve, hard to say if it's more stable, but definitely different issues.  I think moving the external USB WiFi to the Raspberry Pi 4b is working "better", but could be my imagination, so definitely your comment on maybe the 3B+ is under powered with seeing 2.4 and 5ghz channels and numbers of access points / clients connecting to them around me is the culprit of the instability.  Would you like to see the logs of where wlanmon0 is dying, or that's also to be "expected" sometimes?

rsun0525, 2023-08-31:
The "quick" fix when I'm in this state is just to issue " sudo systemctl restart pwnagotchi" and then I'm up and running, is there some watchdog that could look for the error in the logs and run this command to restart it that way?

[2023-08-31 19:26:23,288] [INFO] [ai] ! channels: [1, 4, 7, 9, 10, 11] -> [1, 2, 3, 4, 5, 11]
[2023-08-31 19:26:32,720] [ERROR] Task exception was never retrieved
future: <Task finished name='Task-1' coro=<Client.start_websocket() done, defined at /usr/local/lib/python3.9/dist-packages/pwnagotchi/bettercap.py:40> exception=OSError("Multiple exceptions: [Errno 111] Connect call failed ('::1', 8081, 0, 0), [Errno 111] Connect call failed ('127.0.0.1', 8081)")>
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/dist-packages/pwnagotchi/bettercap.py", line 44, in start_websocket
    async with websockets.connect(s, ping_interval=60, ping_timeout=90) as ws:
  File "/usr/local/lib/python3.9/dist-packages/websockets/legacy/client.py", line 637, in __aenter__
    return await self
  File "/usr/local/lib/python3.9/dist-packages/websockets/legacy/client.py", line 655, in __await_impl_timeout__
    return await self.__await_impl__()
  File "/usr/local/lib/python3.9/dist-packages/websockets/legacy/client.py", line 659, in __await_impl__
    _transport, _protocol = await self._create_connection()
  File "/usr/lib/python3.9/asyncio/base_events.py", line 1064, in create_connection
    raise OSError('Multiple exceptions: {}'.format(
OSError: Multiple exceptions: [Errno 111] Connect call failed ('::1', 8081, 0, 0), [Errno 111] Connect call failed ('127.0.0.1', 8081)
[2023-08-31 19:26:33,396] [ERROR] [agent:_fetch_stats] self.session: ConnectionError(MaxRetryError("HTTPConnectionPool(host='localhost', port=8081): Max retries exceeded with url: /api/session (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x7f49396a90>: Failed to establish a new connection: [Errno 111] Connection refused'))"))
[2023-08-31 19:26:38,464] [ERROR] [agent:_fetch_stats] self.session: ConnectionError(MaxRetryError("HTTPConnectionPool(host='localhost', port=8081): Max retries exceeded with url: /api/session (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x7f493ac370>: Failed to establish a new connection: [Errno 111] Connection refused'))"))
[2023-08-31 19:26:43,512] [ERROR] [agent:_fetch_stats] self.session: ConnectionError(MaxRetryError("HTTPConnectionPool(host='localhost', port=8081): Max retries exceeded with url: /api/session (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x7f481fa400>: Failed to establish a new connection: [Errno 111] Connection refused'))"))
[2023-08-31 19:26:48,564] [ERROR] [agent:_fetch_stats] self.session: ConnectionError(MaxRetryError("HTTPConnectionPool(host='localhost', port=8081): Max retries exceeded with url: /api/session (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x7f481fa490>: Failed to establish a new connection: [Errno 111] Connection refused'))"))
[2023-08-31 19:26:53,652] [ERROR] [agent:_fetch_stats] self.session: ConnectionError(MaxRetryError("HTTPConnectionPool(host='localhost', port=8081): Max retries exceeded with url: /api/session (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x7f48303df0>: Failed to establish a new connection: [Errno 111] Connection refused'))"))
[2023-08-31 19:26:59,476] [ERROR] [agent:_fetch_stats] self.session: ConnectionError(MaxRetryError("HTTPConnectionPool(host='localhost', port=8081): Max retries exceeded with url: /api/session (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x7f481fafd0>: Failed to establish a new connection: [Errno 111] Connection refused'))"))
[2023-08-31 19:27:04,403] [INFO] [epoch 84] duration=00:00:41 slept_for=00:00:41 blind=1 sad=0 bored=0 inactive=1 active=0 peers=0 tot_bond=0.00 avg_bond=0.00 hops=0 missed=0 deauths=0 assocs=0 handshakes=0 cpu=3% mem=50% temperature=45C reward=-0.0058823529411764705
[2023-08-31 19:27:04,405] [INFO] [ai] saving model to /root/brain.nn ...
[2023-08-31 19:27:04,840] [INFO] [ai] saving /root/brain.json

jayofelony, 2023-09-01:
Fix_brcmfmac plugin should be there and enabled.

rsun0525, 2023-09-01:
syntax "error" in my config.toml file, so now it's showing up under plugins, let's see if that solves it.

jayofelony, 2023-09-01:
main.plugins.fix_brcmf_plugin.enabled = true

jayofelony, 2023-09-01:
please run `sudo chmod +x /usr/bin/* && sudo monstart` that may fix some issues with going blind. Thank you.

closed 2023-09-01 by jayofelony