jayofelony/pwnagotchi-bullseye #6: [BUG] FixBRCMF not working?

rsun0525, 2023-09-04:
**Describe the bug**
Installed 2.2.4 on a Raspberry Pi Zero 2W, saw in the log where it said "restarting wifi.recon" but didn't work

**To Reproduce**
Steps to reproduce the behavior:
1.  just leave it there running
2. ...
3. ...

**Expected behavior**
The hope is that FixBRCMF would keep it pwnagotchi running without doing a reboot.

**Screenshots**

[image]

**Environment (please complete the following information):**
 - Pwnagotchi version 2.2.4
 - OS version
 - Type of hardware Raspberry Pi Zero 2 W
[pwnagotchi.log](https://github.com/jayofelony/pwnagotchi/files/12517272/pwnagotchi.log)

 - Any additional hardware used

**Additional context**
Add any other context about the problem here.

rsun0525, 2023-09-05:
Same thing happened to my device that's running a Raspberry Pi 4B, using the built in WiFi adapter.
[pwnagotchi.log](https://github.com/jayofelony/pwnagotchi/files/12517452/pwnagotchi.log)

jayofelony, 2023-09-05:
Yes, version 2.2.7 is up now. This should be fixed and it looks for some extra errors that might occur and tries to automatically fix them.

rsun0525, 2023-09-05:
Thanks for your prompt response.  Manually updated my 3 boxes to 2.2.7 and we'll see how it goes.  I'm assuming you just want the pwnagotchi.log for any troubleshooting, etc.

jayofelony, 2023-09-05:
My PiZ02W has been running almost 3 hours now, without 1 restart of brcmfmac. Seems quite stable somehow.

rsun0525, 2023-09-05:
Since upgrading to 2.2.7, seems like pwnagotchi doesn't honor me setting the blind setting to 10, so it's not autorebooting anymore when there's a problem.
[pwnagotchi.log](https://github.com/jayofelony/pwnagotchi/files/12522470/pwnagotchi.log)

jayofelony, 2023-09-05:
`[2023-09-05 03:28:37,057] [INFO] [epoch 243] duration=00:00:25 slept_for=00:00:24 blind=0 sad=213`

It hasn't gone blind since at least here, didn't scroll back any further. Can you run `sudo aireplay-ng --test wlan0mon` to see if it is still injecting?

rsun0525, 2023-09-05:
pi@pwnagotchi4b:~ $ sudo aireplay-ng --test wlan0mon
Interface wlan0mon:
ioctl(SIOCGIFINDEX) failed: No such device
pi@pwnagotchi4b:~ $ ifconfig
eth0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.191.121  netmask 255.255.255.0  broadcast 192.168.191.255
        inet6 fe80::dea6:32ff:fe30:7490  prefixlen 64  scopeid 0x20<link>
        ether dc:a6:32:30:74:90  txqueuelen 1000  (Ethernet)
        RX packets 126838  bytes 24791782 (23.6 MiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 7268  bytes 831411 (811.9 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        inet6 ::1  prefixlen 128  scopeid 0x10<host>
        loop  txqueuelen 1000  (Local Loopback)
        RX packets 94604  bytes 212136917 (202.3 MiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 94604  bytes 212136917 (202.3 MiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

usb0: flags=4099<UP,BROADCAST,MULTICAST>  mtu 1500
        inet 10.0.0.2  netmask 255.255.255.0  broadcast 10.0.0.255
        ether 66:99:79:37:cd:2c  txqueuelen 1000  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

pi@pwnagotchi4b:~ $

rsun0525, 2023-09-05:
[pwnagotchi.log](https://github.com/jayofelony/pwnagotchi/files/12523276/pwnagotchi.log)

rsun0525, 2023-09-05:
After another reboot, same problem: 

pi@pwnagotchi4b:/var/log $ sudo aireplay-ng --test wlan0mon
Interface wlan0mon:
ioctl(SIOCGIFINDEX) failed: No such device
pi@pwnagotchi4b:/var/log $ ifconfig -a
eth0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.191.121  netmask 255.255.255.0  broadcast 192.168.191.255
        inet6 fe80::dea6:32ff:fe30:7490  prefixlen 64  scopeid 0x20<link>
        ether dc:a6:32:30:74:90  txqueuelen 1000  (Ethernet)
        RX packets 16221  bytes 3221911 (3.0 MiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 808  bytes 137402 (134.1 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        inet6 ::1  prefixlen 128  scopeid 0x10<host>
        loop  txqueuelen 1000  (Local Loopback)
        RX packets 16487  bytes 24600653 (23.4 MiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 16487  bytes 24600653 (23.4 MiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

usb0: flags=4099<UP,BROADCAST,MULTICAST>  mtu 1500
        inet 10.0.0.2  netmask 255.255.255.0  broadcast 10.0.0.255
        ether 02:51:da:67:f4:c8  txqueuelen 1000  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

wlan0: flags=4098<BROADCAST,MULTICAST>  mtu 1500
        ether dc:a6:32:30:74:91  txqueuelen 1000  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

pi@pwnagotchi4b:/var/log $
[pwnagotchi.log](https://github.com/jayofelony/pwnagotchi/files/12523341/pwnagotchi.log)

rsun0525, 2023-09-05:
Running 2.2.8, but still not working exactly right, I would expect it to reboot after 10 based on the blind settings.

[2023-09-05 09:38:14,468] [WARNING] 33 epochs with no activity -> angry

[pwnagotchi.log](https://github.com/jayofelony/pwnagotchi/files/12524235/pwnagotchi.log)

This one is the Raspberry Pi 4B with a Waveshare 3, but using internal WIFi adapter.

jayofelony, 2023-09-05:
I think thats if you use the watchdog plugin, or the restart parameter in config? Not sure though.

jayofelony, 2023-09-05:
Yes, that's bettercap failing. I have a commit on this subject. So stay tuned.

rsun0525, 2023-09-05:
So if I see bettercap failing just reboot at this point until there's another update before uploading more log files right?

rsun0525, 2023-09-06:
When I do an update, it just says 2.2.9, should it say 2.2.9.1?

rsun0525, 2023-09-06:
No worries, just saw 2.2.9.1 show up

jayofelony, 2023-09-06:
I forgot to update the version number so it will remain at 2.2.9. I think there will be a 2.3 today.

rsun0525, 2023-09-07:
With 2.3.0 still seeing the wlan0mon interface die, so right now my only fix is to reboot, is there another way to restart bettercap?  I tried doing a sudo systemctl stop bettercap and the service stopped, but when I started it up again, it didn't start.

rsun0525, 2023-09-07:
[pwnagotchi.log](https://github.com/jayofelony/pwnagotchi/files/12543337/pwnagotchi.log)

rsun0525, 2023-09-07:
[pwnagotchi.log](https://github.com/jayofelony/pwnagotchi/files/12543351/pwnagotchi.log)

rsun0525, 2023-09-07:
[pwnagotchi.log](https://github.com/jayofelony/pwnagotchi/files/12543395/pwnagotchi.log)

rsun0525, 2023-09-07:
Three different Raspberry Pi's running 2.3.0 all quit working and the only way I could figure out how to make it work is to reboot.

jayofelony, 2023-09-07:
It depends if wlan0 is up or not. You could try `sudo monstart`.

rsun0525, 2023-09-07:
sudo monstart gives me this output, but I can now see it in ifconfig, so we'll see if it starts working.

[image]

rsun0525, 2023-09-07:
On the 4 pwnagotchi's I ran it on, 3 of them had wlan0mon show up, while the 4th one, it didn't work, so I rebooted.  Even though the interface showed up on 3 of them the results weren't consistent, I had to do a sudo systemctl restart pwnagotchi and then sometimes it just started working, but when it didn't I just had to do a reboot.

rsun0525, 2023-09-07:
Running 2.3.3 but still wlan0mon dying and can't start it up

[image]

Raspberry Pi Zero 2 W.

[pwnagotchi.log](https://github.com/jayofelony/pwnagotchi/files/12553735/pwnagotchi.log)

It's hit or miss it seems as I have another Raspberry Pi Zero 2 W that so far is still up

rsun0525, 2023-09-07:
same device, Raspberry Pi Zero 2 W, happened again, seems like this isn't happening on the 4B or 3B+ (I found the right now board to try :-)), 
[pwnagotchi.log](https://github.com/jayofelony/pwnagotchi/files/12553791/pwnagotchi.log)

rsun0525, 2023-09-07:
The only other thing to say is that I noticed if the SSID entry is blank, it probably means the wlan0mon died.

[image]

As long as there's a value on the display, it means wlan0mon is still up, but just hope the autoreboot at blind = 10 works, etc.  The pain is really when I'm carrying it around on battery and it's not easy to manually reboot it, but always flipping the switch off and on on the pisugar2 isn't convenient either, etc.

rsun0525, 2023-09-07:
on the 3B+, your change is definitely having an improvement as I see the "fix" working several times: 
[image]

So far I would of expected the 3B+ or 4B to have flacked out by now so it's up and running.

jayofelony, 2023-09-07:
You can try either `sudo ifconfig wlan0 up && sudo monstart` or just `sudo monstart`. I'm trying to catch all breaks of the firmware. But for now try to start it manually.

jayofelony, 2023-09-07:
I just figured out that the display-password is a faulty plugin, whatever you do, do not install it. Messes up the pwnagotchi somehow.

rsun0525, 2023-09-08:
No worries, I don't use that plugin.
[image]

rsun0525, 2023-09-08:
> You can try either `sudo ifconfig wlan0 up && sudo monstart` or just `sudo monstart`. I'm trying to catch all breaks of the firmware. But for now try to start it manually.

[image]

What is happening, it's killing the WIFi adaptor so badly only a reboot solves it.

rsun0525, 2023-09-08:
[image]

rsun0525, 2023-09-08:
So if it sees the WiFi adaptor "crash", force a reboot?  I'm guessing there's no newer BCM43430/2 driver available to add?

jayofelony, 2023-09-08:
Yes, I will change fix_services to reboot when this happens.

Nope, no newer drivers available.

jayofelony, 2023-09-08:
I will close this now, since fix_services picks up most of the nexmon driver errors and tries to automatically fix them. THere may still be some left unresolved, you can open a new one for those.

closed 2023-09-08 by jayofelony