jayofelony/pwnagotchi-bullseye #7: [BUG]

rsun0525, 2023-09-05:
Not sure what this is, but AI can't start on Raspberry Pi Zero 2W since going to 2.2.8

I deleted brain.*, rebooted, and still get this error (but it did start up properly on the 4B following the same procedure).

[2023-09-05 08:50:24,537] [INFO] [epoch 1] duration=00:00:24 slept_for=00:00:22 blind=0 sad=0 bored=0 inactive=0 active=1 peers=2 tot_bond=0.32 avg_bond=0.16 hops=1 missed=0 deauths=0 assocs=2 handshakes=0 cpu=25% mem=60% temperature=66C reward=0.10071428571428571
[2023-09-05 08:50:25,263] [INFO] [ai] creating model ...
[2023-09-05 08:50:25,265] [ERROR] error while starting AI (__init__() got an unexpected keyword argument 'alpha')
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/dist-packages/pwnagotchi/ai/__init__.py", line 63, in load
    a2c = A2C(SB_A2C_POLICY, env, **config['params'])
TypeError: __init__() got an unexpected keyword argument 'alpha'
[2023-09-05 08:50:25,268] [WARNING] [ai] AI not loaded!
[2023-09-05 08:50:26,996] [INFO] new peer pwnagotchix@dd303b616b42dea814aafc29aaad588de5425050107f3f41e4844b4c17f0c3a5 detected (1912 encounters)
[2023-09-05 08:50:30,045] [INFO] new peer pwnagotchi400@34d7114159b654b1212497f7cbc29e8ab2f64aa8c4c6ac47f936c3653dbde29e detected (1239 encounters)
[2023-09-05 08:50:47,250] [INFO] sending association frame to <hidden> (80:30:dc:17:0f:53 Texas Instruments) on channel 1 [0 clients], -70 dBm...
[2023-09-05 08:50:47,361] [INFO] sending association frame to <hidden> (80:30:dc:16:fd:11 Texas Instruments) on channel 1 [0 clients], -72 dBm...
[2023-09-05 08:50:47,422] [INFO] [epoch 2] duration=00:00:22 slept_for=00:00:22 blind=0 sad=0 bored=0 inactive=0 active=2 peers=4 tot_bond=0.47 avg_bond=0.12 hops=1 missed=0 deauths=0 assocs=2 handshakes=0 cpu=10% mem=70% temperature=63C reward=0.13404761904761905

jayofelony, 2023-09-05:
You have a deprecated parameter in your config.toml. Delete all ai parameters from your config please.

rsun0525, 2023-09-05:
Done, delete everything with ai.params in config.toml, deleted the logs, rebooted, and now starting fresh.

closed 2023-09-05 by rsun0525