jayofelony/pwnagotchi-bullseye #8: [BUG] Raspberry Pi 3B+ only sees 2.4ghz channels?

rsun0525, 2023-09-06:
Running 2.2.9.1 on a Raspberry Pi 3B+ with internal Wi-Fi adapter

Looking in the /var/log/pwnagotchi.log, I see this:

[INFO] supported channels: [1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11]

Would I expect to see 5ghz channels as well?

On my Raspberry Pi 4B with internal Wi-Fi adapter I see this:

[INFO] supported channels: [1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 36, 40, 44, 48, 52, 56, 60, 64, 100, 104, 108, 112, 116, 120, 124, 128, 132, 136, 140, 144, 149]

While on a Raspberry Pi Zero 2W with internal Wi-Fi adapter I see this:

 [INFO] supported channels: [1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11]

Maybe just need to compile a newer drive for 3B+?

jayofelony, 2023-09-06:
It should do so automatically, so either you have those channels disabled or it doesn't support 5ghz for whatever reason.

What's the output of `dmesg | grep brcm`

do-ki, 2023-09-06:
Verify it it is B+ raspberry Pi 3B only have 2.4Ghz

rsun0525, 2023-09-06:
If you look at the specifications, it lists 5ghz?  https://www.raspberrypi.com/products/raspberry-pi-3-model-b-plus/

rsun0525, 2023-09-06:
pi@pwnagotchi3b:~ $ dmesg | grep brcm
[   10.091350] brcmfmac: loading out-of-tree module taints kernel.
[   10.091361] brcmfmac: loading out-of-tree module taints kernel.
[   10.150874] brcmfmac: F1 signature read @0x18000000=0x1541a9a6
[   10.164480] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43430-sdio for chip BCM43430/1
[   10.165459] usbcore: registered new interface driver brcmfmac
[   10.861386] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43430-sdio for chip BCM43430/1
[   10.890399] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43430-sdio for chip BCM43430/1
[   10.890611] brcmfmac: brcmf_c_process_clm_blob: no clm_blob available (err=-2), device may have limited channels available
[   10.895418] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM43430/1 wl0: Aug  7 2017 00:46:29 version 7.45.41.46 (r666254 CY) FWID 01-f8a78378
[   17.043395] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save disabled
[   18.181387] brcmfmac: brcmf_vif_add_validate: Attempt to add a MONITOR interface...
[   18.181423] brcmfmac: brcmf_mon_add_vif: brcmf_mon_add_vif called
[   18.181434] brcmfmac: brcmf_mon_add_vif: Adding vif "wlan0mon"

jayofelony, 2023-09-06:
That's not a 3b+, it uses the 43455 wifi chip. You have the 43430 wifi chip. That does not support 5ghz.

closed 2023-09-06 by jayofelony

rsun0525, 2023-09-06:
My fault, I had thought this was a B+, but you are correct, I put on the latest Raspberry Pi OS and it's only a B.