jayofelony/pwnagotchi-bullseye #9: [BUG] After upgrading to 2.3, can't get to web ui on port 8080

rsun0525, 2023-09-06:
To confirm it wasn't my config.toml, I removed it and only using the default.toml file and still won't start.

[pwnagotchi.log](https://github.com/jayofelony/pwnagotchi/files/12538763/pwnagotchi.log)

Should we also stop this service as well?
 pwngrid-peer.service

renewedbullet, 2023-09-06:
same issue talked with xbits about it already, did a brand new flash on my pi4 and stopped working after going to 2.3

jayofelony, 2023-09-06:
Well the strange thing is I don't have this issue at all. I wouldn't know where to begin looking, since the logs don't show anything error related.

Lenorr, 2023-09-06:
I think i found the Issue. Pwngrid fails to start, because it is not marked as an executable. This makes pwngrid-peer.service and pwngrid.service fail, which results in pwnagotchi.service restarting endlessly. Running `sudo chmod +x /usr/local/bin/pwngrid` and rebooting fixed it for me.
[image]

jayofelony, 2023-09-06:
Awesome

rsun0525, 2023-09-06:
after changing the permissions, I'm up, but still bettercap not stable, will add my question to the other bug

rsun0525, 2023-09-07:
I'm assuming this can be closed as the fix is to make sure /usr/local/bin/pwngrid is executable, at least going to 2.3.3 the problem didn't come up, so I'm assuming if I reimage and update manually to the latest one, it'll work.

closed 2023-09-07 by rsun0525