jayofelony/pwnagotchi-bullseye #10: [BUG] "blind" count doesn't seem to increment in 2.3.5?

rsun0525, 2023-09-09:
Running 2.3.5 on a Raspberry Pi Zero 2W, I saw the Wi-Fi driver "crash".

[image]

Would expect pwnagotchi to start incrementing blind count, so it would auto reboot itself, etc.

[pwnagotchi.log](https://github.com/jayofelony/pwnagotchi/files/12565565/pwnagotchi.log)

rsun0525, 2023-09-09:
On a Raspberry Pi 4B, I had the wlan0mon die, but wlan0 was still up, doing a "sudo monstart" didn't work, and got this error:

[image]

But your code must of figured out it was down because eventually it rebooted itself automatically, so didn't give me a chance to dig deeper if I could of manually got it working, etc.  so this is definitely the expected outcome when there's a problem so that it can start working again with me manually rebooting it.

rsun0525, 2023-09-09:
On a Raspberry pi 3B, I also had wlan0mon die, but wlan0 was still up, doing a "sudo monstart" didn't work, and got this error:

[image]

Now I'm letting it sit hoping your scripts detect it's still down because I couldn't manually fix it, and auto reboot itself.

[pwnagotchi.log](https://github.com/jayofelony/pwnagotchi/files/12565599/pwnagotchi.log)

rsun0525, 2023-09-09:
This was the screen I continuously saw on the Raspberry Pi 3B+ and 4B, but I'm thinking this is being captured in the pwnagotchi.log so you can see it there:

[image]

jayofelony, 2023-09-09:
> Running 2.3.5 on a Raspberry Pi Zero 2W, I saw the Wi-Fi driver "crash".
> 
> [image]
> 
> Would expect pwnagotchi to start incrementing blind count, so it would auto reboot itself based on this setting:
> 
> main.mon_max_blind_epochs = 50
> 
> [pwnagotchi.log](https://github.com/jayofelony/pwnagotchi/files/12565565/pwnagotchi.log)

No, because firmware crashed completely. So it doesn't even have eyes anymore to be blind with.

> On a Raspberry pi 3B+, I also had wlan0mon die, but wlan0 was still up, doing a "sudo monstart" didn't work, and got this error:
>
> [image]
>
> Now I'm letting it sit hoping your scripts detect it's [...]

Depends on what error is given, and where as well.

> This was the screen I continuously saw on the Raspberry Pi 3B+ and 4B, but I'm thinking this is being captured in the pwnagotchi.log so you can see it there:
> 
> [image]

This is a bettercap issue, it is ddos'ing itself. Nothing I can do about it, atleast for now.

rsun0525, 2023-09-10:
Running 2.3.6, and both my Raspberry Pi Zero 2 W's WiFi adapters still crashed, so wasn't sure if you wanted to see the logs from both if you can glean something from it.

[pwnagotchiz0.log](https://github.com/jayofelony/pwnagotchi/files/12569537/pwnagotchiz0.log)
[pwnagotchiz2.log](https://github.com/jayofelony/pwnagotchi/files/12569538/pwnagotchiz2.log)

closed 2023-09-17 by jayofelony