jayofelony/pwnagotchi-bullseye #15: [BUG] Brain create fails

rohanday3, 2023-09-20:
**Describe the bug**
AI Brain parameter error

**To Reproduce**
Steps to reproduce the behavior:
1. Follow instructions
2. Update to 2.4.2

**Expected behavior**
Brain gets created

**Environment (please complete the following information):**
 - Pwnagotchi version 2.4.2
 - Type of hardware Pi0W 2

**Additional context**
16:20:56 [ERROR] error while starting AI (__init__() got an unexpected keyword argument 'alpha')
(most recent call last):
 File "/usr/local/lib/python3.9/dist-packages/pwnagotchi/ai/__init__.py", line 62, in load
   a2c = A2C(SB_A2C_POLICY, env, **config['params'])
__init__() got an unexpected keyword argument 'alpha'
16:20:56 [INFO] Deleting brain and restarting.
16:20:57 [WARNING] [ai] AI not loaded!

jayofelony, 2023-09-20:
You are using an old version of config.toml. Delete all ai parameters in your config!

closed 2023-09-20 by jayofelony

rohanday3, 2023-09-21:
My bad, thanks for answering my question so quickly. Love the great job with reviving the repo. I want to create a script to port the config file as i see there's already one to fix the wifi driver issue.