jayofelony/pwnagotchi-bullseye #18: [BUG] AI not training properly

jayofelony, 2023-10-08:
```LOG
/usr/local/lib/python3.9/dist-packages/stable_baselines3/common/vec_env/base_vec_env.py:234: UserWarning: You tried to call render() but no `render_mode` was passed to the env constructor.
  warnings.warn("You tried to call render() but no `render_mode` was passed to the env constructor.")
INFO:root:[AI] learning for 50 epochs ...
[2023-10-08 10:48:09,557] [INFO] [AI] learning for 50 epochs ...
ERROR:root:[AI] error while training (could not create a primitive descriptor for a matmul primitive)
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/dist-packages/pwnagotchi/ai/train.py", line 185, in _ai_worker
    self._model.learn(total_timesteps=epochs_per_episode, callback=self.on_ai_training_step)
  File "/usr/local/lib/python3.9/dist-packages/stable_baselines3/a2c/a2c.py", line 194, in learn
    return super().learn(
  File "/usr/local/lib/python3.9/dist-packages/stable_baselines3/common/on_policy_algorithm.py", line 259, in learn
    continue_training = self.collect_rollouts(self.env, callback, self.rollout_buffer, n_rollout_steps=self.n_steps)
  File "/usr/local/lib/python3.9/dist-packages/stable_baselines3/common/on_policy_algorithm.py", line 169, in collect_rollouts
    actions, values, log_probs = self.policy(obs_tensor)
  File "/usr/local/lib/python3.9/dist-packages/torch/nn/modules/module.py", line 1518, in _wrapped_call_impl
    return self._call_impl(*args, **kwargs)
  File "/usr/local/lib/python3.9/dist-packages/torch/nn/modules/module.py", line 1527, in _call_impl
    return forward_call(*args, **kwargs)
  File "/usr/local/lib/python3.9/dist-packages/stable_baselines3/common/policies.py", line 625, in forward
    values = self.value_net(latent_vf)
  File "/usr/local/lib/python3.9/dist-packages/torch/nn/modules/module.py", line 1518, in _wrapped_call_impl
    return self._call_impl(*args, **kwargs)
  File "/usr/local/lib/python3.9/dist-packages/torch/nn/modules/module.py", line 1527, in _call_impl
    return forward_call(*args, **kwargs)
  File "/usr/local/lib/python3.9/dist-packages/torch/nn/modules/linear.py", line 114, in forward
    return F.linear(input, self.weight, self.bias)
RuntimeError: could not create a primitive descriptor for a matmul primitive
```

If anyone with some AI experience could please provide some explanation it would be much appreciated. I have tried everything I could think of. But to no avail.

jayofelony, 2023-10-10:
Downgrade torch and tochvision fixes this error for now.

`sudo pip3 install torch==2.0.1 torchvision==0.15.2`

NikoKamtsioras, 2023-10-11:
how do i install them?

NikoKamtsioras, 2023-10-11:
> how do i install them? my Pwnagotchi does not have wifi.
nvm sry!

jayofelony, 2023-10-13:
You got them installed?

NikoKamtsioras, 2023-10-13:
yep!

jayofelony, 2023-10-16:
Permanent fix in coming update, downgrade to latest working known version of torch and torchvision.

closed 2023-10-16 by jayofelony