jayofelony/pwnagotchi-bullseye #22: "default route" not set on a 3B+, but works just fine a 4B and 400 [BUG]

rsun0525, 2023-10-17:
**Describe the bug**
Installed latest 2.4.8 image on Raspberry Pi 3B+, 4B, and 400, all get their IP address through DHCP on the Ethernet port, but for some odd reason, the default route doesn't get set on a 3B+, but it does get set on a 4B and 400

**To Reproduce**
Steps to reproduce the behavior:
Install 2.4.8 on a Raspberry Pi 3B+, plug in the Ethernet, do a "netstat -r", mysteriously 3B+ doesn't have default route, but do the exact same thing on a 4B or 400 and it does get set.

**Expected behavior**
Expect the 3B+ to get the default route

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Environment (please complete the following information):**
 - Pwnagotchi version 2.4.8
 - OS version Linux pwnagotchi3b 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux
 - Type of hardware Raspberry Pi 3B+

**Additional context**
As a work around, I added "sudo route add default gw xx.xx.xx.xx" to the /usr/bin/pwnagotchi-launcher so that the default route would get set each time a reboot occured (of course I can't easily move this pwnagotchi to a different network now).

jayofelony, 2023-10-17:
What is output of `route -n` on the pi3?

rsun0525, 2023-10-18:
***@***.***:~ $ route -n
Kernel IP routing table
Destination     Gateway         Genmask         Flags Metric Ref    Use
Iface
0.0.0.0         10.0.0.1        0.0.0.0         UG    203    0        0 usb0
10.0.0.0        0.0.0.0         255.255.255.0   U     0      0        0 usb0
192.168.191.0   0.0.0.0         255.255.255.0   U     203    0        0 eth0

Doing the exact same install on a 4b, the route is just fine:

***@***.***:~ $ route -n
Kernel IP routing table
Destination     Gateway         Genmask         Flags Metric Ref    Use
Iface
0.0.0.0         192.168.191.1   0.0.0.0         UG    202    0        0 eth0
0.0.0.0         10.0.0.1        0.0.0.0         UG    203    0        0 usb0
10.0.0.0        0.0.0.0         255.255.255.0   U     0      0        0 usb0
192.168.191.0   0.0.0.0         255.255.255.0   U     202    0        0 eth0

jayofelony, 2023-10-18:
Somehow the pi3 doesnt have usb0, it gets metric 0. Which is prioritised over eth0 which has metric 203. I'll dive into it.

jayofelony, 2023-10-18:
Okay I think I have finally found a permanent fix for all the networking issues!!

closed 2023-10-18 by jayofelony