jayofelony/pwnagotchi-bullseye #24: After installing 2.5.0 Raspberry Pi 3B+ doesn't work anymore [BUG]

rsun0525, 2023-10-20:
**Describe the bug**
Ran 2.4.9 on a Raspberry Pi 3B+, seem to work well enough, but after reimaging to 2.5.0, 3B+ won't start up anymore

**To Reproduce**
Steps to reproduce the behavior:
1. Installed 2.5.0

**Expected behavior**
Expect it just to work

**Screenshots**
See attachment

**Environment (please complete the following information):**
 - Pwnagotchi version 2.5.0
 - OS version Linux pwnagotchi3b 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux
 - Type of hardware Raspberry Pi 3B+

[pwnagotchi.log](https://github.com/jayofelony/pwnagotchi/files/13056636/pwnagotchi.log)

renewedbullet, 2023-10-20:
sudo rfkill unblock all

rsun0525, 2023-10-20:
That solved it, how to automate this so I don't have to run this command
every time it restarts?

rsun0525, 2023-10-20:
> we were are working on a fix should be in the next 2.5 release.

No worries, we can assume it's closed then since it'll be resolved in the
next version.

renewedbullet, 2023-10-20:
Yeah it can't close them but I had the issue with it this morning on the 02 and pi4 so we reverted back to 2.4.9 since 02 worked fine on that one, well do more investigating as soon as possible.

jayofelony, 2023-10-20:
`sudo rfkill unblock all`

You only have to do this command once, after first boot.

closed 2023-10-20 by jayofelony