jayofelony/pwnagotchi-bullseye #33: [BUG] RPI 4 64bit 8gb reboots during update to 2.5.1

Maynardnaze, 2023-11-12:
When updating from 2.5.0 to 2.5.1, RPI will reboot.

Steps to reproduce the behavior:

1. sudo pwnagotchi --check-update
2. wait a few minutes, will reboot
3. update fails

**Expected behavior**

**Environment (please complete the following information):**
 - Pwnagotchi version 2.5.1
 - OS version 64bit
 - Type of hardware RPI 4 8gb
 - Any additional hardware used External Wifi. Fails with onboard as we..

jayofelony, 2023-11-12:
Any logs would be appreciated, otherwise there is not much I can do. Or screenshots perhaps?

jayofelony, 2023-11-17:
@Maynardnaze please respond or I will close this as solved.

Maynardnaze, 2023-11-17:
Go ahead and close it. 

I'll try a fresh 2.5.2 install.

Logs looked normal from pwnlog. update successfull, then reboot.

Sorry for the delayed response!

 
 

Sent: Friday, November 17, 2023 at 1:49 PM

ly0koS, 2023-11-18:
The same thing happened when updating from 2.5.1 to 2.5.2

To make it more clear:

- Use command `sudo pwnagotchi --check-update`
- Confirm update 
- Check `pwnlog`
- Got the following:
```
05:10:43 [INFO] [update] checking for updates ...
05:11:11 [WARNING] update for jayofelony/pwnagotchi available (local version is '2.5.1'): https://github.com/jayofelony/pwnagotchi/archive/v2.5.2.zip
05:11:11 [INFO] [update] turned off wifi recon ...
05:11:11 [INFO] [update] downloading https://github.com/jayofelony/pwnagotchi/archive/v2.5.2.zip to /usr/local/src/pwnagotchi/pwnagotchi_2.5.2.zip ...
05:11:26 [INFO] [update] extracting /usr/local/src/pwnagotchi/pwnagotchi_2.5.2.zip to /usr/local/src/pwnagotchi ...
05:11:27 [INFO] [update] installing pwnagotchi ...
05:11:27 [INFO] [update] done
05:11:27 [WARNING] rebooting ...
05:11:37 [WARNING] syncing...
``` 

- Rpi rebooting
- Version still be 2.5.1 at the front screen and `sudo pwnagotchi --version`
- There is the .zip file inside `/usr/local/src/pwnagotchi/`
- But that's all and no other files found inside that folder

jayofelony, 2023-11-18:
All files get deleted after installing if I'm correct.

jayofelony, 2023-11-18:
`sudo nano /usr/local/lib/python3.9/dist-packages/pwnagotchi/plugins/default/auto-update.py`

CTRL +W: jayofelony/pwnagotchi

Edit `jayofelony/pwnagotchi` to `jayofelony/pwnagotchi-torch` CTRL+X, Y, Enter

`sudo reboot`

Than after you boot you run `sudo pwnagotchi --check-update` normally.

closed 2023-11-19 by Maynardnaze