jayofelony/pwnagotchi-bullseye #43: [BUG] Fail to install on RPI 4 8GB

Maynardnaze, 2023-11-21:
**Describe the bug**

Fresh download and clean install, no modifications.
I'ved tried 2.5.1, 2.5.2 and 2.5.3
RPI 4 8GB
I've tried copying a working config.toml and placing in boot.

Steps to reproduce the behavior:
1. Download
2. Flash
3. Wont boot, No SSH access

Should there be anything in config.toml for a clean fresh install?

jayofelony, 2023-11-21:
Did you download the correct image, check if it has rpi in the name. If you downloaded the opi, orange pi, it wont work.

Maynardnaze, 2023-11-21:
No go on a fresh download. using arm64 image. 
I have not tried 2.5.2 though. 
Worth giving it a shot?

jayofelony, 2023-11-21:
Almost done with 2.5.4, will upload shortly. But I feel like it's a you problem, not a me problem. If you get me. Maybe power issues, the images are all bootable.

Maynardnaze, 2023-11-21:
@jayofelony You probably arent wrong!
I'll try some spare naked RPI 4s when I get home.
This PI has a GPS and X728 UPS hat.
For future reference, should there be a config.toml in /boot on first start?

Maynardnaze, 2023-11-21:
Do you have the hash values by any chance?

jayofelony, 2023-11-21:
> @jayofelony You probably arent wrong! I'll try some spare naked RPI 4s when I get home. This PI has a GPS and X728 UPS hat. For future reference, should there be a config.toml in /boot on first start?

No. There is a dummy config.toml in /etc/pwnagotchi created in the build.

jayofelony, 2023-11-21:
> Do you have the hash values by any chance?

No, only download from my repo. I am the only one that is allowed to create releases.

jayofelony, 2023-11-21:
https://github.com/jayofelony/pwnagotchi-torch/releases/tag/v2.5.4

closed 2023-11-21 by jayofelony

jayofelony, 2023-11-22:
Whatever you like buddy!

Maynardnaze, 2023-12-01:
It was the SD-Card. lol always the dumb shit.