jayofelony/pwnagotchi-bullseye #55: Adding the ability to easily connect to wifi networks using the internal wifi chip, when using external adapter to deauth

beeinaboats, 2023-12-10:
**Is your feature request related to a problem? Please describe.**
Problem, having to always get the pwnagotchi to connect to bluetooth tether with my phone to get internet.

**Describe the solution you'd like**
Having the pwnagotchi connect to wifi networks listed in a config option. 

I know you can't have the pwnagotchi deauth if the wifi chip is having to be connected to a network, however, many people, like myself, use an external wifi adapter to deauth, and just turn the pi's wifi chip off. What I'm wanting is to instead of turning the chip off, using it to connect to networks that we list. I know what I'm wanting is possible and can be done individually, but it requires a lot of changes to be done, which is why most people don't do it. Having the ability to just click a button in the config, and type our network and password would be awesome 

**Additional context**
this just an idea, ok if it nots something that you'd want to take on

jayofelony, 2023-12-11:
I like the idea, I don't know if it's feasible. I'll just keep it open in case any one feels up for it.

kr4k0n, 2023-12-20:
I like the idea too and that's what I did at times, although, personally, it defeats the purpose of being portable on a RPI or Orange Pi like [mine](https://www.tindie.com/products/pisugar/pwnagotchi-complete-pack-pi0w-eink-battery-case/) unless I have it installed on my Asus ROG Ally or GPD Win 3 and 4.

Previously thought of a plugin till I noticed that linux would mistakenly switch out the interface names if external adapter are left plugged in before boot.

JD-2006, 2024-01-04:
I am interested in this also. I will dig into some old school projects like pwnpi n pwniexpress for some inspiration. Will report back with any updates. Some scripts for 'Evil Twin AP' might help.

JD-2006, 2024-01-08:
Well funny enough, or not funny after 10s of hours testing and googling...there's this lol
https://github.com/unagisan69/pwnagotchi-external-wifi-plugin

The adapter naming convention needs to be changed before use. Will test today.

@jayofelony  Do you think this would work?

jayofelony, 2024-01-08:
Well the problem is more that Linux has the tendency to randomly name a wireless adapter wlan0, when using more than 1 after boot.

jayofelony, 2024-01-08:
fix_services would do that

JD-2006, 2024-01-08:
Ok cool thx.
Preciate your guidance.

If I start the Pi with an external usb wifi plugged in, even after a while it never powers the adapter.
But if I plug it in after booting it powers and recognizes the adapter.

Any clue on that one?

jayofelony, 2024-01-09:
Not of the top of my head

mdstricklin, 2024-01-10:
Not sure, but this documentation may be helpful.
https://wiki.debian.org/NetworkInterfaceNames

JD-2006, 2024-01-10:
Thanks. Will add to growing list of ways I can piss off my OS. Ha..
Then it will be a pile of networking services the were never meant to run.

JD-2006, 2024-01-10:
Some slight successes. Hardware maybe 80% there.... 
bettercap-launcher[977]: [22:14:34] [sys.log] [inf] wifi using interface wlan1mon (0)

Some route issues with bt-tether with dongle plugged in.

mdstricklin, 2024-01-10:
Ahhh... I wonder if that's why I'm having bt-tether issues on mine!

JD-2006, 2024-01-11:
Also though, I wired a GPS module and shutdown switch to my GPIO, and added 
enable_uart=1
dtoverlay=miniuart-bt to config.txt so my GPS could use the main uart. So who knows what perils I have created for myself.

JD-2006, 2024-01-12:
@beeinaboats  :)
Some promising news to share. Unfortunately "easily connect to wifi" is the opposite of what I have come up with.
I will post a link to my repo when I clean up the instructions n code a bit. Since there are tons of hardware configs out there, it will just have to be a rabbit hole of a project.

On a Pi Zero w with a TP-Link TL-WN722N and a rando GPS module from a drone project. I have...
- The internal wlan0 connecting to a wifi access point with internet
- External wlan1 is getting put into monitor mode
- Bluetooth tethering (sometimes working)
- GPS working

There's so many caveats though. Like updates will break most of your hard work (Thanks @jayofelony ).
But some fiddling would bring it back to working order.