jayofelony/pwnagotchi-bullseye #56: [BUG] Bluetooth randomly keeps crashing on Rpi 02w

kr4k0n, 2023-12-12:
**Describe the bug**

Bluetooth has a notch to keep crashing and dying connected to my phone (Fold 4 Samsung). 

**To Reproduce**

Steps to reproduce the behavior:
1. Pair and trust phone via bluetoothctl
2. Add mac address in config.toml
3. Just let gotchi sit through out and wait for the crash to happened. Fastest to crash is 5mins tops. Longest run is an hour.

**Screenshots**

[image]
[image]
[image]
[image]

**Environment (please complete the following information):**

```
Version	2.5.4
Build	Pwnagotchi-Torch by Jayofelony
uname	Linux Phr34kz 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr 3 17:24:16 BST 2023 aarch64 GNU/Linux
```
[https://opwngrid.xyz/search/c937ecfddb8ec5d71b6a4cc700cb90a8ba35bcf5f098f9c5059a8f4de433c7df](https://opwngrid.xyz/search/c937ecfddb8ec5d71b6a4cc700cb90a8ba35bcf5f098f9c5059a8f4de433c7df)

 - Type of hardware: 02w
 - Any additional hardware used: PiSugar 2 (2 led), Waveshare v4 and Samsung Galaxy Fold 4 (Android 13)

**Additional context**

Noticed - "a2dp-sink profile connect failed for" in systemctl status bluetooth

did
```
sudo apt-get install pulseaudio-module-bluetooth
sudo killall pulseaudio
pulseaudio --start    
sudo systemctl restart bluetooth
```
and added `load-module module-bluez5-discover` in `/etc/pulse/default.pa` to no avail but it did prolong the crash period together with --noplugin=sap as stated in [here](https://raspberrypi.stackexchange.com/questions/40839/sap-error-on-bluetooth-service-status). Bluetooth audio slider on bluetooth settings in mobile have to manually be turned on.

----------------

Reference:
https://askubuntu.com/questions/1172000/a2dp-sink-profile-connect-failed
https://raspberrypi.stackexchange.com/questions/40839/sap-error-on-bluetooth-service-status

-----------------
Others: 

- Changed DNS from 1.1.1.1 to 8.8.4.4 as well in [https://github.com/jayofelony/pwnagotchi-torch/blob/pwnagotchi-torch/builder/data/etc/dhcpcd.conf](https://github.com/jayofelony/pwnagotchi-torch/blob/pwnagotchi-torch/builder/data/etc/dhcpcd.conf). 1.1.1.1 DNS is a slow DNS for me.

- Changed DNS from 9.9.9.9 to 8.8.8.8 in [https://github.com/jayofelony/pwnagotchi-torch/blob/pwnagotchi-torch/pwnagotchi/plugins/default/bt-tether.py](https://github.com/jayofelony/pwnagotchi-torch/blob/pwnagotchi-torch/pwnagotchi/plugins/default/bt-tether.py). Quad 9 DNS is a slow DNS for me as well.

- Changed

```
#self.set('friend_face', peer.face())
self.set('friend_face', ' (⌐■_■)')            
```

in [https://github.com/jayofelony/pwnagotchi-torch/blob/pwnagotchi-torch/pwnagotchi/ui/view.py](https://github.com/jayofelony/pwnagotchi-torch/blob/pwnagotchi-torch/pwnagotchi/ui/view.py) so as peer face does not appear as /CUSTOM-FACES/LOOK-R.PNG on both device and web ui

- Changed

```
def _on_face_change(self, old, new):
self._advertisement['face'] = '(⌐■_■)'
grid.set_advertisement_data(self._advertisement)
```

in [https://github.com/jayofelony/pwnagotchi-torch/blob/pwnagotchi-torch/pwnagotchi/mesh/utils.py](https://github.com/jayofelony/pwnagotchi-torch/blob/pwnagotchi-torch/pwnagotchi/mesh/utils.py) so as peer face does not appear as /CUSTOM-FACES/LOOK-R.PNG in https://opwngrid.xyz/

kr4k0n, 2023-12-12:
Not sure if this could relate to current issue at hand
https://github.com/lategoodbye/rpi-zero/commit/f7ad01a69fd1e319354912f926b9accd59757b24#diff-df21c16c23d5cdf50e8f79d9c579a3a605ad022314038818f10db1fbffb68b0d

jayofelony, 2023-12-13:
The custom faces is being taken care of on opwngrid.xyz. So I don't need changes in my code, as it would affect all faces. i could changr the friend face as you mentioned.

I'll look into the rest, I do know bt is a bitch to put it mildly.

jayofelony, 2023-12-13:
SAP is disabled by default in the image build, as rpi has no sim to access internet data.

https://github.com/jayofelony/pwnagotchi-torch/blob/2de631c40953a534b91a2354b01850e4a38c4e09/builder/raspberrypi64.yml#L181C26-L181C26

kr4k0n, 2023-12-13:
> The custom faces is being taken care of on opwngrid.xyz. So I don't need changes in my code, as it would affect all faces. i could changr the friend face as you mentioned.
> 
> I'll look into the rest, I do know bt is a bitch to put it mildly.

Yeah. BT is. I noticed leaving the device overnight would easily overheat the board till 80c degrees from 50c. When comparing to 1.5.5 by evilsocket, I am still clueless of what went wrong

jayofelony, 2023-12-13:
Can you post your custom config as well for me, the bt part.

kr4k0n, 2023-12-13:
> Can you post your custom config as well for me, the bt part.

Here it is. I used the same strings on 0w as well.

[image]

jayofelony, 2023-12-13:
> > Can you post your custom config as well for me, the bt part.
> 
> Here it is. I used the same strings on 0w as well. Only changed the "IP address" part to .45 on 02w if I need them both turned on at the same time.
> 
> [image]
> 

Yeah cool 😎, seems fine.

kr4k0n, 2023-12-20:
Using [Watchdog](https://github.com/evilsocket/pwnagotchi/blob/master/pwnagotchi/plugins/default/watchdog.py) plugin temporarily fixes the issue for the time being. To reboot instead of a restart.

jayofelony, 2023-12-27:
In version 2.5.8 I have done some fixes to BT mistakes I made when building 2.5.4 image. Can you see how it goes with this new version @kr4k0n ?

kr4k0n, 2023-12-28:
> In version 2.5.8 I have done some fixes to BT mistakes I made when building 2.5.4 image. Can you see how it goes with this new version @kr4k0n ?
> 
> Edit Oh and run `sudo systemctl enable bluetooth.service` if you will please.

YAY! Thank you @jayofelony ! In the process I got 2 4gb orangepi 02w as well. Shall do an tests on raspi 02w as well and report back

jayofelony, 2023-12-28:
> YAY! Thank you @jayofelony ! In the process I got 2 4gb orangepi 02w as well. Shall do an tests on raspi 02w as well and report back

The opi isnt really suitable, it's chinese made. No built-in wifi monitor mode support. Very difficult to build drivers for external antenna.

closed 2024-01-10 by jayofelony