jayofelony/pwnagotchi-bullseye #63: [BUG] Not seeing nearby pwnagotchi

jfiggins, 2023-12-15:
**Describe the bug**
Flashed a RPi 3b and got it up and running. I currently have a ZeroW that is up and running as well running 1.5.5. Both have personality.advertise = true. The ZeroW is seeing my 3b pwnagotchi, but the 3b is not seeing the ZeroW.

**To Reproduce**
Steps to reproduce the behavior:
1. ZeroW running 1.5.5
2. RPi 3b running 2.5.6
3. Power up both devices, wait for them to search for eachother

**Expected behavior**
Both devices see eachother, not just the ZeroW on 1.5.5 seeing the RPi 3b running 2.5.6

**Screenshots**
[image]
[image]

**Environment (please complete the following information):**
 - Pwnagotchi version 2.5.6

**Additional context**
Add any other context about the problem here.

jayofelony, 2023-12-15:
Eh, funny. I never had 2 pwns together, let alone with different images. But I do know they have to be on exactly the same channel to see eachother. How have they been powered on?

jfiggins, 2023-12-15:
I haven't had them on super long. Probably 10 minutes or so. Going to leave them both on for about an hour to see if it makes any difference. Then I'm going to give flashing the ZeroW with the latest version of torch to see if that fixes it or not. Just weird that the Zero is seeing the 3b but the 3b isn't seeing the Zero, and they have virtually the same config file. If getting them running the same image doesn't fix it, it seems like `/mesh/peer.py` or `/mesh/utils.py` would be the place to look/mess with to get it working? I could always give tweaking that a shot and seeing if I could fix it that way too.

jfiggins, 2023-12-15:
So I did some digging, and also updated my ZeroW to run torch. Neither device was picking up the other. I readded the removed lines from peer.py in this commit to my ZeroW: https://github.com/jayofelony/pwnagotchi-torch/commit/fae58c8f24864d3dc651e45c83a37f21287644b1#diff-355be70d854d5145b69e3a06dfa743bd0bc987c1be291318793d5dc58eebeff7

And now the name of the 3b is showing, but no face. It seems like some of the changes in this commit broke the peer feature. I didn't change anything on the 3b, and it's not showing the ZeroW.

jfiggins, 2023-12-15:
Hrm, the 3b is seeing the Zero now, but missing the faces that are originally in the firmware.

jayofelony, 2023-12-16:
Yeah, i commented out the faces because people use the custom faces mod. And that doesnt work on peer faces as not everyone have it installed. Even if you have 2, you might not use it on both devices.

closed 2023-12-20 by jfiggins