jayofelony/pwnagotchi-bullseye #68: invert colors via config.toml

mdstricklin, 2024-01-02:
**Is your feature request related to a problem? Please describe.**
Enhancement request. Presently, users have to invert the color presentation by editing view.py, which appears to be overwritten or reverted under certain conditions (including updates)

**Describe the solution you'd like**
Adding a parameter to the config.toml file might provide a better way to preserve user settings between updates. Perhaps adding something like ui.display.invertcolors would be a good way to address the issue.

jayofelony, 2024-01-04:
I have tried several implementations but config cannot be called from that point in view.py sadly. So I am unable to implement this.

jayofelony, 2024-01-04:
I am trying to get this working, bare with me.

closed 2024-01-05 by jayofelony