jayofelony/pwnagotchi-bullseye #69: Add buttons under Plugins for update, install, upgrade

JD-2006, 2024-01-03:
**Describe the solution you'd like**

Adding buttons in the webui for update, install, upgrade of plugins.

jayofelony, 2024-01-04:
I like this idea!

jayofelony, 2024-01-04:
@JD-2006 
[image]
Something like this?

JD-2006, 2024-01-04:
Yeah exactly, or a single button for upgrading. Whichever is easier. Was thinking of the plugins cli options to buttons. Me thinks the 'list' one would be too cluttered? If your on your phone adding plugins/plugin use from putty seems painful in the cli. Thanks

jayofelony, 2024-01-04:
Yeah, installing is not possible right now. I got the upgrade button working. In the background it will run update and upgrade of that particular plugin you clicked upgrade for.

closed 2024-01-04 by jayofelony

JD-2006, 2024-01-04:
Cheers thx!