jayofelony/pwnagotchi-bullseye #75: [BUG] 2.5.4 BT-Tether is not working

travka, 2024-01-09:
**Describe the bug**
Hi. I'm trying configure 2.5.4 version to get tethered by bluetooth via android to update but nothing happens, it simply doesn't  even trying to find the phone, and i don't understand where i should investigate because there's no logs regarding bluetooth

so configuration in config.toml is:

`main.plugins.bt-tether.enabled = true
main.plugins.bt-tether.devices.android-phone.enabled = true
main.plugins.bt-tether.devices.android-phone.search_order = 1
main.plugins.bt-tether.devices.android-phone.mac = "XX:XX:XX:XX:XX:XX"
main.plugins.bt-tether.devices.android-phone.ip = "192.168.44.44"
main.plugins.bt-tether.devices.android-phone.netmask = 24
main.plugins.bt-tether.devices.android-phone.interval = 1
main.plugins.bt-tether.devices.android-phone.scantime = 5   
main.plugins.bt-tether.devices.android-phone.max_tries = 999
main.plugins.bt-tether.devices.android-phone.share_internet = false
main.plugins.bt-tether.devices.android-phone.priority = 1
`

**Environment**
 - 2.5.4
 - Android 13

Kal-Toh, 2024-01-09:
Have you tried connecting manually? 
You can do this by using `sudo bluetoothctl` to enter the Bluetooth shell. You can use `scan on` from the shell while your phone is discoverable until it picks it up. Once picked up use `scan off` to stop scanning. 

To connect use `pair <mac address>` (at this point you should get a notification on your phone, follow instructions if any in shell) and save trust the device with `trust <mac address>` 

Exit out of the bt shell, reboot and you should be connected.

travka, 2024-01-09:
Can you give me an advice how to do thar ? Because i've tried connect external keyboard via second usb port, but without success

Kal-Toh, 2024-01-10:
If you're on Windows you can use Putty to connect to it while's it's plugged into your PC via the data port (inner microUSB port). Have a go at following the guide here:
https://pwnagotchi.ai/configuration/#connect-to-your-pwnagotchi

jayofelony, 2024-01-10:
Set share_internet to true and also set up bt tether to enabled on your phone. It's under mobile hotspot or something. Than use bluetoothctl as mentioned before.

NikoKamtsioras, 2024-01-11:
Having the same problem, have tried the above, but it is still not working. Using release 2.6.3.

jayofelony, 2024-01-11:
What worked for me yesterday, was simply turning off BT on your phone and turning it back on, than turn on bt tethering on the phone. Than did the bluetoothctl again, but first I removed the device in bluetoothctl. And removed my pwnagotchi from phone memory.

NikoKamtsioras, 2024-01-12:
Still not working for me, tried that several times

NikoKamtsioras, 2024-01-12:
My android phone does not give me a prompt.

NikoKamtsioras, 2024-01-12:
NOTE: if you have a pi2w use bookworm

oakgul1, 2024-01-12:
I also got the same issue, i manually paired the phone and from the pwnagotchi. But i have the pi0w so bookworm dont know if it is possible. i first use the pwnagotchi-raspberrypi-os-lite-v1.5.6-beta2 but i dont know why to problem persist. is there any other thing i can do to resolve this ?

jayofelony, 2024-01-12:
Bluetooth and RPi are a bad mix it seems. It just a game of luck apparently.

JD-2006, 2024-02-18:
Here's a workaround that works for my pi0w. After you have paired and trusted your phone. 

- Install bluez-test-scripts:
sudo apt-get install --no-install-recommends bluez-test-scripts

sudo cp /usr/share/doc/bluez-test-scripts/examples/bluezutils.py /usr/bin
sudo cp /usr/share/doc/bluez-test-scripts/examples/test-network /usr/bin

Create a system service.
sudo nano /etc/systemd/system/auto-bt-pan-connect.service
- Use your phones MAC instead of xx:xx:xx:xx:xx:xx

[Unit]
Description=Auto PAN Connect Service
After=usb-gadget.target 

[Service]
Type=oneshot
ExecStart=/usr/bin/python3 test-network xx:xx:xx:xx:xx:xx nap
WorkingDirectory=/usr/bin
StandardOutput=inherit
StandardError=inherit
Restart=on-failure
RestartSec=90
User=pi

[Install]
WantedBy=multi-user.target

- Save and exit.

sudo systemctl daemon-reload
sudo systemctl start auto-bt-pan-connect

- Enable the service to start on boot:
sudo systemctl enable auto-bt-pan-connect

I have it try to connect every 1 1/2 minutes so it doesn't fill up the logs because it will create an error when it runs and there's no phone to connect to.