jayofelony/pwnagotchi-bullseye #77: [BUG] [v2.6.1]waveshare2in13b_v4 flickers between black and white

kagehisa, 2024-01-10:
**Describe the bug**
Whenever the screen is refreshed the screen flickers between a deep black background and the inverted state. Also setting it to permanently white background is not possible.
I used the setting `ui.display.color = "black"` in my `config.toml`, sice this also seems to be the default. Here is the pwnlog entry for this setting:

```
14:29:18 [INFO] Session-stats plugin loaded.
14:29:18 [INFO] webcfg: Plugin loaded.
14:29:18 [INFO] [Fix_Services] plugin loaded.
14:29:18 [INFO] grid plugin loaded.
14:29:18 [INFO] [update] plugin loaded.
14:29:19 [WARNING] ui.fps is 0, the display will only update for major changes
14:29:19 [INFO] initializing waveshare 2.13inb v2in13_V4 display
14:29:19 [INFO] Got here
14:29:28 [INFO] pwni@de7691f9de29100ddc2383231f434ce52164982df54bb2b9013ff7251d3fa4af (v2.6.1)
14:29:28 [INFO] entering manual mode ...
14:29:29 [INFO] the last session lasted 21 minutes, 35 seconds (0 completed epochs, trained for 0), average reward:0.0 (min:1000 max:-1000)
14:29:29 [INFO] web ui available at http://[::]:8080/
```

When I try to change the color to white in `config.toml`, I get an error about a wrong image size in the pwnlog and the display is permanently flickering between black and white background without showing any pwnagotchi stats or faces. This is what the pwnlog locks like with the setting set to white:

```
14:37:58 [INFO] Session-stats plugin loaded.
14:37:58 [INFO] [update] plugin loaded.
14:37:58 [INFO] webcfg: Plugin loaded.
14:37:58 [INFO] grid plugin loaded.
14:37:58 [INFO] [Fix_Services] plugin loaded.
14:37:59 [WARNING] ui.fps is 0, the display will only update for major changes
14:37:59 [INFO] initializing waveshare 2.13inb v2in13_V4 display
14:37:59 [INFO] Got here
14:38:08 [WARNING] Wrong image dimensions: must be 122x250
14:38:08 [WARNING] Wrong image dimensions: must be 122x250
14:38:08 [INFO] pwni@de7691f9de29100ddc2383231f434ce52164982df54bb2b9013ff7251d3fa4af (v2.6.1)
14:38:08 [INFO] entering manual mode ...
14:38:09 [INFO] web ui available at http://[::]:8080/
14:38:09 [INFO] the last session lasted 30 minutes, 15 seconds (0 completed epochs, trained for 0), average reward:0.0 (min:1000 max:-1000)
14:38:10 [WARNING] Wrong image dimensions: must be 122x250
14:38:10 [WARNING] Wrong image dimensions: must be 122x250
14:38:16 [WARNING] Wrong image dimensions: must be 122x250
14:38:16 [WARNING] Wrong image dimensions: must be 122x250
14:38:21 [WARNING] Wrong image dimensions: must be 122x250
14:38:21 [WARNING] Wrong image dimensions: must be 122x250
14:38:24 [WARNING] Wrong image dimensions: must be 122x250
14:38:24 [WARNING] Wrong image dimensions: must be 122x250
14:38:28 [WARNING] Wrong image dimensions: must be 122x250
14:38:28 [WARNING] Wrong image dimensions: must be 122x250
14:38:30 [WARNING] Wrong image dimensions: must be 122x250
14:38:30 [WARNING] Wrong image dimensions: must be 122x250
14:38:34 [WARNING] Wrong image dimensions: must be 122x250
14:38:34 [WARNING] Wrong image dimensions: must be 122x250
14:38:36 [WARNING] Wrong image dimensions: must be 122x250
14:38:36 [WARNING] Wrong image dimensions: must be 122x250
14:38:40 [WARNING] Wrong image dimensions: must be 122x250
14:38:40 [WARNING] Wrong image dimensions: must be 122x250
14:38:43 [WARNING] Wrong image dimensions: must be 122x250
14:38:43 [WARNING] Wrong image dimensions: must be 122x250
14:38:46 [WARNING] Wrong image dimensions: must be 122x250
14:38:46 [WARNING] Wrong image dimensions: must be 122x250
14:38:49 [WARNING] Wrong image dimensions: must be 122x250
14:38:49 [WARNING] Wrong image dimensions: must be 122x250
14:38:53 [WARNING] Wrong image dimensions: must be 122x250
14:38:53 [WARNING] Wrong image dimensions: must be 122x250
14:38:56 [WARNING] Wrong image dimensions: must be 122x250
14:38:56 [WARNING] Wrong image dimensions: must be 122x250
14:38:59 [WARNING] Wrong image dimensions: must be 122x250
14:38:59 [WARNING] Wrong image dimensions: must be 122x250
```

Am I doing something wrong? 

**To Reproduce**
Steps to reproduce the behavior:
1. Install pwnagotchi-torch and update to latest version 2.6.1
2. Include the following lines in your 
```
ui.display.enabled = true
ui.display.type = "waveshare213inb_v4"
ui.display.color = "white"
```
3. restart pwnagotchi service
4. Enjoy the lightshow 

**Expected behavior**
I would like to either set a white background with black drawlines or a black background with white drawlines. If either is selected I would like it to not constantly flicker between the color modes.

**Environment (please complete the following information):**
 - v2.6.1
 - Raspian OS Buster
 - Raspberry Pi 0W
 - Waveshare 2.13 e-Paper display V4

A1fi, 2024-01-10:
Cd /usr/local/lib/
Ls should be pyton3.9 on latest img just check
So run 
Sudo nano /usr/local/lib/pyton3.9/dist-package/pwnagotchi/ui/view.py
And invert the colours 
Ur config should say black and waveshare_3 
Hope this helps

A1fi, 2024-01-10:
Also join discord for help

A1fi, 2024-01-10:
If you're plugged into data port it will be in MANUal mode
If u plug into power port it will switch to AUTO and get more screen refreshes and should be okay

A1fi, 2024-01-10:
You can also use default config change ur setting using webcfg and it will write everything in config properly with no mistakes

kagehisa, 2024-01-10:
Ok, thank you for your help. Guess this can be closed then.

closed 2024-01-10 by kagehisa