jayofelony/pwnagotchi-bullseye #81: [BUG] v2.6.3 pwnagotchi commands respond with 'unexpected intend' error

kagehisa, 2024-01-11:
**Describe the bug**
Since update to 2.6.3 each call to `pwnagotchi` results in the following error message:
```
Traceback (most recent call last):
  File "/usr/local/bin/pwnagotchi", line 12, in <module>
    from pwnagotchi import utils
  File "/usr/local/lib/python3.9/dist-packages/pwnagotchi/utils.py", line 168
    for boot_conf in ['/boot/config.yml', '/boot/firmware/config.yml', '/boot/config.toml', '/boot/firmware/config.toml']:
IndentationError: unexpected indent
``` 

The pwnagotchi service keeps on restarting due to this.
pwnlog is quiet in regards to this.

**To Reproduce**
Steps to reproduce the behavior:
1. Fresh image with versin 2.5.4, update to 2.6.3
2. try any `sudo pwnagotchi` command

**Expected behavior**
Working pwnagotchi.

NikoKamtsioras, 2024-01-11:
https://github.com/jayofelony/pwnagotchi-torch/commit/55ce23900128d9698b8a27803518e37afb5a23aa

jayofelony has made a commit to fix the issue.

kagehisa, 2024-01-11:
Did just that, thanks.

closed 2024-01-11 by kagehisa

NikoKamtsioras, 2024-01-11:
No problem!